umrover/mrover-ros2 #49: Receive CAN FD messages on STM32 Nucleo

owenpark8, 2024-11-15:
Use fdcanusb from laptop to send message to NUCLEO-G431RB with CAN FD transceiver (ISO1044DEVM). CAN bus should be at 1 Mbps nominal / 5 Mbps data. Should be able to see message reception on Nucleo with breakpoint on HAL_FDCAN_RxFifo0Callback().

artyom-boyarov, 2024-11-22:
### Nov 21
Sent messages from one nucleo to another. The boards were connected to CAN transceivers and with a proper CAN bus (so CANH/CANL lines)

artyom-boyarov, 2024-12-04:
## Dec 3
Successfully received messages sent through FDCAN-USB. 
### Setup
#### Sender laptop side
Connect FDCAN-USB to CAN transceiver. Wires are black = ground, green = low, blue = high. Connect transceiver VCC to 5V on an STM32 plugged into the **sender** laptop. Run the following commands:
`sudo ip link add name can1 type vcan`
(in new terminal)
`mrover`
`sudo ./ansible/roles/esw/files/bin/fdcanusb_daemon -F -v /dev/fdcanusb can1`
To send messages run:
`cansend <HEX ID>##<1-digit HEX flags><message>`
e.g.
`cansend 011##01234`

#### Receiver laptop side
Plug in STm32 and connect TX to transceiver TX, RX to transceiver RX, ditto for 5V and GND. Run.

Works in both bus monitoring and normal mode.

closed 2024-12-04 by artyom-boyarov