umrover/mrover-ros2 #50: Investigate CANalyzer interface options

owenpark8, 2024-11-15:
Screen: Size, interface (SPI, LTDC, etc), touch screen?
Buttons: What will user input look like? How many buttons? Keyboard? Etc.

artyom-boyarov, 2024-11-22:
### Sample displays
https://www.buydisplay.com/9-inch-lcd-module-tft-display-w-touch-panel-i2c-serial-spi-arduino - 9" display capable of I2C interface.

### Note: only the following boards support to LTDC controller for interfacing with LCD displays:
• STM32F429/439 and STM32F469/479 lines
• STM32F7x6, STM32F7x7, STM32F7x8, and STM32F7x9 lines
• STM32H7A3/B3, STM32H742, STM32H743/753, STM32H745/755, STM32H747/757 lines
• STM32H7B0, STM32H750 value lines
• STM32H7R3/7S3 and STM32H7R7/7S7 lines
• STM32L4+ series or STM32L4R/S and STM32L4P/Q
• STM32U595/5A5, STM32U599/5A9, STM32U5F7/5G7, STM32U5F9/5G9 lines
If we want to use the NUCLEO-G431RBs we need to use an I2C/SPI-based LCD display.

### Code examples for LTDC controller:
https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/STM32H747I-DISCO/Examples/LCD_DSI

artyom-boyarov, 2024-12-04:
https://www.adafruit.com/product/2050
https://www.adafruit.com/product/2090
https://www.adafruit.com/product/1591
https://www.adafruit.com/product/1680
https://www.adafruit.com/product/1028

closed 2026-01-17 by jbrhm